compas-dev/compas #1273: `MissingMemberException` when trying to extend `SceneObject`

chenkasirer, 2024-01-24:
Been trying to create a new `SceneObject` implementation for something in COMPAS Timber and got this issue where it expects whatever item it wraps to have a `name` attribute. 

I think the culprit is this line: 
https://github.com/compas-dev/compas/blob/5bcf73733262230cc543505daf78270688afd5fd/src/compas/scene/sceneobject.py#L71

It seems it's supposed to take it from `kwargs`, from the instance or use the class name if the other two don't work. But really if the instance doesn't have `name` it always breaks.

Admittedly, I had to ask ChatGPT why this doesn't work cause for the life of me I couldn't figure it out, apparently the stuff around the `or` always gets evaluated..

@Licini maybe `getattr(item, "name", item.__class__.__name__)` instead?

**To Reproduce**
Steps to reproduce the behavior:
1. Inherit from `SceneObject`
2. Pass something that doesn't have attribute `name` as `item`.

```
Runtime error (MissingMemberException): 'FeatureApplicationError' object has no attribute 'name'

Traceback:
  line 25, in __init__, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas_rhino\scene\sceneobject.py"
  line 73, in __init__, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\scene\sceneobject.py"
  line 15, in __init__, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas_timber\rhino\scene\featureerrorobject.py"
  line 262, in add, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\scene\scene.py"
  line 11, in script
```

**Expected behavior**
Not assume `name` from `item`

**Desktop (please complete the following information):**
 - Rhino / GH

tomvanmele, 2024-01-25:
@chenkasirer if the item is a data class, how come it doesn't have a name attribute?

chenkasirer, 2024-01-25:
Does it have to be a `Data`? In my case I'm just trying to implement visualization of some custom errors for visual debugging in GH.

tomvanmele, 2024-01-25:
the entire setup is based on pairing `compas.data.Data` objects with `compas.scene.SceneObject` objects, so kind of :)

tomvanmele, 2024-01-25:
otherwise perhaps just use `compas_rhino.conversions` directly? this actually works well in GH...

chenkasirer, 2024-01-25:
Gotcha. `Data` it is then..

chenkasirer, 2024-01-25:
> otherwise perhaps just use `compas_rhino.conversions` directly? this actually works well in GH...

it's a container for several (different) geometries and was thinking of a text label as well, but still figuring stuff out with the new `Scene`.

tomvanmele, 2024-01-25:
let me know if you need help...

Licini, 2024-01-26:
If it's ok I'm closing this now, as our plan will be #1276