compas-dev/compas #1282: Networkx not compatible with Py3.7

yck011522, 2024-01-30:
I am running some tests on a branch of compas_fab that will make it more* compatible with compas 2.0.
I came across an error `ERROR: No matching distribution found for networkx>=3.0` for the environment `build-cpython (macos-py37)`

https://github.com/compas-dev/compas_fab/actions/runs/7710486107/job/21013894034?pr=400

I found a recent commit in `compas` that stopped it testing against 3.7, @Licini 
https://github.com/compas-dev/compas/pull/1160

I would like to know if the longer-term plan is to remove support for Python3.7 completely (and to keep Networkx in requirement), or if we plan to go back to supporting 3.7 (and we use an older Networkx)?

tomvanmele, 2024-01-30:
https://devguide.python.org/versions/

python 3.7 has reached end of life mid last year...

yck011522, 2024-01-31:
So, is compas officially dropping support for 3.7? 

If so, should compas_robot and compas_fab follow? @gonzalocasas 

And I would suggest updating this:
https://github.com/compas-dev/compas/blob/b812aeada3a307924b18cbf5e8afa02ae4775dc0/setup.py#L43

gonzalocasas, 2024-01-31:
3.7 is end-of-life, so yes, let's drop it across the board

jf---, 2024-04-06:
@tomvanmele || @yck011522 plz close

closed 2024-04-07 by yck011522