compas-dev/compas #1287: Add __str__ to all geometry classes for pretty print

yck011522, 2024-01-31:
# Feature Request

I found the pretty-print `__str__` function added by @gonzalocasas in (https://github.com/compas-dev/compas/pull/1234) helps a lot when making example files. Should we add this to other geometry classes?

I can help make a PR if we are okay with this. @tomvanmele

tomvanmele, 2024-02-01:
@yck011522 that would be great!

yck011522, 2024-02-02:
I wonder if the pretty print should include the attribute labels or not:
@tomvanmele @gonzalocasas 

`Point(1.678, 2.344, 5.754)` or  `Point(x=1.678, y=2.344, y=5.754)` 

At the moment, even the `__repr__` functions are not consistent. For example, `Point.__repr__()` - Has the x,y,z while `Quaternion__repr__()` does not.

yck011522, 2024-02-02:
@gonzalocasas @tomvanmele 
Do we need some tests for these __str__ methods? I ran into problems (those broken code parts) when I tested them in a console, so maybe a simple test to make sure the str() function actually runs, for example:
```
test___str__():
   p = Polyhedron.from_platonicsolid(12)
   str(p)
```

gonzalocasas, 2024-02-02:
As for the proposal of not including attributes in str, I actually think the new way with attribute labels makes it more readable and despite the fact this means more updates to docs and tests in other packages, I think we should keep them.

PR #1289 (2024-07-10): Add some more __str__ function to geometry classes for pretty print
Partially Fixes #1287 

### What type of change is this?

- [ ] Bug fix in a **backwards-compatible** manner.
- [x] New feature in a **backwards-compatible** manner.
- [ ] Breaking change: bug fix or new feature that involve incompatible API changes.
- [ ] Other (e.g. doc update, configuration, etc)

### Checklist

_Put an `x` in the boxes that apply. You can also fill these out after creating the PR. If you're unsure about any of them, don't hesitate to ask. We're here to help! This is simply a reminder of what we are going t [... cut]
closed 2024-07-10 by yck011522