compas-dev/compas #1310: Color Error when the inputs are partly `int`

ZacZhangzhuo, 2024-03-13:
**Describe the bug**
As titled.

**To Reproduce**
```python
from compas.colors import Color

color = Color(0.0,0.0,0)

print(Color.coerce(color))
```

``bash
File "\compas\src\compas\colors\color.py", line 585, in coerce
    raise ColorError
compas.colors.color.ColorError
```

jf---, 2024-04-06:
@tomvanmele  the current implementation coerces `str` `int` and `float` as an argument.
this screams for a refactor with [`typing.overload`](https://pub.towardsai.net/the-python-decorator-that-supercharges-developer-experience-78b3fe7f1682) which was introduced in py 3.5

tomvanmele, 2024-04-08:
that would indeed be very nice, but unfortunately, COMPAS 2.x still has to maintain compatibility with IronPython. once we start COMPAS 3 (which will be soon), we can start using this kind of mechanism...

jf---, 2024-04-08:
Terrific, could you pls label it as `compas_3`?

tomvanmele, 2024-04-08:
sure done