compas-dev/compas #1322: scipy >= 1.1 requirement - windows is able to install only through conda

petrasvestartas, 2024-04-12:
**Describe the bug**

I have an issue only on WINDOWS, on UNIX everything works, related to SCIPY 
when typing to   `pip install -r requirements.txt` and  `pip install -r requirements-dev.txt`.

Description: 
`Scipy package failed due to missing Fortran compilers, which are required for building some of the components of SciPy. The log indicates that attempts to find various Fortran compilers (ifort, gfortran, flang, pgfortran, and g95) on your system were unsuccessful.`

Error is gone when using conda: `conda install scipy`. 

**To Reproduce**
Steps to reproduce the behavior:
1. windows laptop and cookiecutter requirements

**Expected behavior**
Working pip instead of conda

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Python version=3.8.16
 - Python package manager [pip]

jf---, 2024-04-12:
Looks like you don't have a Fortran compiler installed. A shame that SciPy doesn't offer binary wheels packages. Perhaps being more explicit regarding supporting pip || conda is a good idea. I'd advocate for conda as well, since it's a reliable way of installing compas related dependencies like OCC, CGAL, libgl and ROS for example.

tomvanmele, 2024-04-16:
i see that your environment is based on Python 3.8. just want to point out again that Python 3.8 has reached end-of-life already last year.

in any case, we can not do anything about official releases of other packages. at best we could point out this problem on the issue tracker of the package in question...

gonzalocasas, 2024-04-16:
I can confirm that 3.9 on Windows works for a `pip` install. I'm going to go ahead and close it, feel free to keep the discussion going if anyone disagrees with the decision.

closed 2024-04-16 by gonzalocasas