compas-dev/compas #1324: compas 2 installation error on Mac

begums, 2024-04-16:
**Describe the bug**
I come across the error below, when I try to install compas_fab with compas 2 with Rhino 8 on Mac. We solved this issue on Windows by installing Microsoft [Build Tools for Visual Studio](https://visualstudio.microsoft.com/downloads/#build-tools-for-visual-studio-2022), however not sure how to proceed in Mac. 

**To Reproduce**
Steps to reproduce the behavior:
1. I successfully installed compas_fab with:
```
(conda_env) python -m compas_fab
Yay! COMPAS FAB is installed correctly! 
```
2. Then I try to install compas_fab on Rhino by running this line on terminal:
```
(conda_env) /Users/username/.rhinocode/py39-rh8/python3.9 -m pip install compas_fab
```
3. See error:
```
Collecting compas_fab
  Using cached compas_fab-1.0.2-py2.py3-none-any.whl.metadata (5.3 kB)
Requirement already satisfied: compas<3,>=2.0.4 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from compas_fab) (2.0.4)
Collecting compas-robots<1,>=0.3 (from compas_fab)
  Using cached compas_robots-0.3.0-py2.py3-none-any.whl.metadata (2.7 kB)
Requirement already satisfied: roslibpy>=1.1.0 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from compas_fab) (1.6.0)
Collecting pybullet (from compas_fab)
  Using cached pybullet-3.2.6.tar.gz (80.5 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: pyserial in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from compas_fab) (3.5)
Requirement already satisfied: jsonschema in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from compas<3,>=2.0.4->compas_fab) (4.21.1)
Requirement already satisfied: networkx>=3.0 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from compas<3,>=2.0.4->compas_fab) (3.2.1)
Requirement already satisfied: numpy>=1.15.4 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from compas<3,>=2.0.4->compas_fab) (1.26.4)
Requirement already satisfied: scipy>=1.1 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from compas<3,>=2.0.4->compas_fab) (1.12.0)
Requirement already satisfied: typing-extensions in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from compas<3,>=2.0.4->compas_fab) (4.10.0)
Requirement already satisfied: watchdog in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from compas<3,>=2.0.4->compas_fab) (4.0.0)
Requirement already satisfied: autobahn>=17.10 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from roslibpy>=1.1.0->compas_fab) (23.6.2)
Requirement already satisfied: twisted>=17.9 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from twisted[tls]>=17.9->roslibpy>=1.1.0->compas_fab) (23.10.0)
Requirement already satisfied: txaio>=21.2.1 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from autobahn>=17.10->roslibpy>=1.1.0->compas_fab) (23.1.1)
Requirement already satisfied: cryptography>=3.4.6 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from autobahn>=17.10->roslibpy>=1.1.0->compas_fab) (42.0.5)
Requirement already satisfied: hyperlink>=21.0.0 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from autobahn>=17.10->roslibpy>=1.1.0->compas_fab) (21.0.0)
Requirement already satisfied: setuptools in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from autobahn>=17.10->roslibpy>=1.1.0->compas_fab) (58.1.0)
Requirement already satisfied: attrs>=21.3.0 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from twisted>=17.9->twisted[tls]>=17.9->roslibpy>=1.1.0->compas_fab) (23.2.0)
Requirement already satisfied: automat>=0.8.0 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from twisted>=17.9->twisted[tls]>=17.9->roslibpy>=1.1.0->compas_fab) (22.10.0)
Requirement already satisfied: constantly>=15.1 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from twisted>=17.9->twisted[tls]>=17.9->roslibpy>=1.1.0->compas_fab) (23.10.4)
Requirement already satisfied: incremental>=22.10.0 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from twisted>=17.9->twisted[tls]>=17.9->roslibpy>=1.1.0->compas_fab) (22.10.0)
Requirement already satisfied: zope-interface>=5 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from twisted>=17.9->twisted[tls]>=17.9->roslibpy>=1.1.0->compas_fab) (6.2)
Requirement already satisfied: idna>=2.4 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from twisted[tls]>=17.9->roslibpy>=1.1.0->compas_fab) (3.6)
Requirement already satisfied: pyopenssl>=21.0.0 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from twisted[tls]>=17.9->roslibpy>=1.1.0->compas_fab) (24.0.0)
Requirement already satisfied: service-identity>=18.1.0 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from twisted[tls]>=17.9->roslibpy>=1.1.0->compas_fab) (24.1.0)
Requirement already satisfied: jsonschema-specifications>=2023.03.6 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from jsonschema->compas<3,>=2.0.4->compas_fab) (2023.12.1)
Requirement already satisfied: referencing>=0.28.4 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from jsonschema->compas<3,>=2.0.4->compas_fab) (0.33.0)
Requirement already satisfied: rpds-py>=0.7.1 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from jsonschema->compas<3,>=2.0.4->compas_fab) (0.18.0)
Requirement already satisfied: six in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from automat>=0.8.0->twisted>=17.9->twisted[tls]>=17.9->roslibpy>=1.1.0->compas_fab) (1.16.0)
Requirement already satisfied: cffi>=1.12 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from cryptography>=3.4.6->autobahn>=17.10->roslibpy>=1.1.0->compas_fab) (1.16.0)
Requirement already satisfied: pyasn1 in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from service-identity>=18.1.0->twisted[tls]>=17.9->roslibpy>=1.1.0->compas_fab) (0.5.1)
Requirement already satisfied: pyasn1-modules in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from service-identity>=18.1.0->twisted[tls]>=17.9->roslibpy>=1.1.0->compas_fab) (0.3.0)
Requirement already satisfied: pycparser in ./.rhinocode/py39-rh8/lib/python3.9/site-packages (from cffi>=1.12->cryptography>=3.4.6->autobahn>=17.10->roslibpy>=1.1.0->compas_fab) (2.21)
Using cached compas_fab-1.0.2-py2.py3-none-any.whl (5.0 MB)
Using cached compas_robots-0.3.0-py2.py3-none-any.whl (2.7 MB)
Building wheels for collected packages: pybullet
  Building wheel for pybullet (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for pybullet (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [3622 lines of output]
      macosx-10.9-universal2
      numpy is disabled. getCameraImage maybe slower.
      darwin!
      found resource files: 3568
      ....
      packages
      ['pybullet_examples', 'pybullet_envs', 'pybullet_data', 'pybullet_robots', 'pybullet_utils', 'pybullet_envs.stable_baselines', 'pybullet_envs.minitaur', 'pybullet_envs.prediction', 'pybullet_envs.agents', 'pybullet_envs.deep_mimic', 'pybullet_envs.examples', 'pybullet_envs.bullet', 'pybullet_envs.baselines', 'pybullet_envs.minitaur.envs_v2', 'pybullet_envs.minitaur.agents', 'pybullet_envs.minitaur.vision', 'pybullet_envs.minitaur.actuatornet', 'pybullet_envs.minitaur.robots', 'pybullet_envs.minitaur.envs', 'pybullet_envs.minitaur.envs_v2.sensors', 'pybullet_envs.minitaur.envs_v2.scenes', 'pybullet_envs.minitaur.envs_v2.tasks', 'pybullet_envs.minitaur.envs_v2.utilities', 'pybullet_envs.minitaur.envs_v2.examples', 'pybullet_envs.minitaur.envs_v2.evaluation', 'pybullet_envs.minitaur.envs_v2.env_wrappers', 'pybullet_envs.minitaur.agents.baseline_controller', 'pybullet_envs.minitaur.agents.tools', 'pybullet_envs.minitaur.agents.scripts', 'pybullet_envs.minitaur.agents.ppo', 'pybullet_envs.minitaur.agents.trajectory_generator', 'pybullet_envs.minitaur.robots.safety', 'pybullet_envs.minitaur.robots.utilities', 'pybullet_envs.minitaur.robots.safety.python', 'pybullet_envs.minitaur.envs.env_randomizers', 'pybullet_envs.agents.tools', 'pybullet_envs.agents.ppo', 'pybullet_envs.deep_mimic.learning', 'pybullet_envs.deep_mimic.gym_env', 'pybullet_envs.deep_mimic.env', 'pybullet_envs.deep_mimic.learning.solvers', 'pybullet_envs.deep_mimic.learning.nets', 'pybullet_data.configs_v2', 'pybullet_data.testdata', 'pybullet_data.policies', 'pybullet_data.configs', 'pybullet_data.configs_v2.wrappers', 'pybullet_data.configs_v2.scenes', 'pybullet_data.configs_v2.tasks', 'pybullet_data.configs_v2.robots', 'pybullet_data.configs_v2.base', 'pybullet_data.testdata.test_imu_state_estimator [... cut ...]
      xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
       ....
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pybullet
Failed to build pybullet
ERROR: Could not build wheels for pybullet, which is required to install pyproject.toml-based projects
```

**Desktop (please complete the following information):**
 - OS: Mac iOS
 - compas 2 with Rhino 8
 - Python package manager: pip, conda

tomvanmele, 2024-04-16:
hi,

please move this to the `compas_fab` issue tracker...

begums, 2024-04-17:
> hi,
> 
> please move this to the `compas_fab` issue tracker...

I cannot move the issue as I am not authorized to do so, but I will make a new issue there and close this one.

closed 2024-04-17 by begums

gonzalocasas, 2024-04-22:
https://github.com/compas-dev/compas_fab/issues/415

tomvanmele, 2024-04-23:
@gonzalocasas something you think we need to do on the compas side related to this?

gonzalocasas, 2024-04-23:
Not much, this is an issue with one of the dependencies of `compas_fab` failing to pip install. But tangentially related, I wonder if we should try to make `python -m compas_rhino.install -v 8.0` also trigger the `install_with_pip` stuff so that "one (install) command rules them all"?

tomvanmele, 2024-04-23:
would be cool, but i haven't been able to get this command to work...

tomvanmele, 2024-04-23:
actually, tried it again because i didn't remember what the problem was at the time and seems to work at lease in the following cases
* `python -m compas_rhino.install_with_pip "-e ."`
* `python -m vompas_rhino.install_with_pip requests`

tomvanmele, 2024-04-23:
it failed somewhere but don't remember where. i also wrote it a while back when quite a few things with the CPython interpreter didn't really work yet...