compas-dev/compas #1351: componentizer default install path is wrong for Rhino 8

obucklin, 2024-05-02:
`python -m compas_rhino.install` adds the COMPAS packages to `~/.rhinocode/py27-rh8/Lib/site-packages` instead of `\AppData\Roaming\McNeel\Rhinoceros\8.0\scripts`. Grasshopper components cannot then import the modules.  

there is a workaround which is to include 

`pt = r"C:\Users\obucklin/.rhinocode/py27-rh8/Lib/site-packages"

if pt not in sys.path:
    sys.path.append(pt)`

at the top of a python component. This fixes IronPython components.

I'm not sure if it's the same problem, but cpython components built with `invoke build-cpython-ghuser-components` do not instantiate on the canvas and result in a `User Object failed to deserialize` error:

[image]

Windows 11
Rhino 8 SR6 2024-4-10
COMPAS 2.1.0

Thanks!

petrasvestartas, 2024-05-02:
I guess ScriptEditor usage for Rhino8 is still experimental as stated in the Compas documentation:
[image]
https://github.com/compas-dev/compas/blob/08f06247cfdd99c4fdb029a4d74430ca0efed444/src/compas_rhino/install.py#L35
Can it be that python library installation directory and Grasshopper are two different directories?

What works for me is following:

ScriptEditor Rhino8 CPython on windows:
`python src\rhino\gh\componentize_cpy.py src\rhino\gh\cpy %APPDATA%\Grasshopper\UserObjects`

on mac
`python src/rhino/gh/componentize_cpy.py src/rhino/gh/cpy  "/Users/petras/Library/Application Support/McNeel/Rhinoceros/8.0/Plug-ins/Grasshopper (b45a29b1-4343-4035-989e-044e8580d9cf)/UserObjects"`

or the package manager directory (manual upload to Rhino).

The python libraries I install as described in compas main page:
https://compas.dev/compas/latest/userguide/cad.rhino8.html
When a library is ready, I do it via pip, since this is the only possible solution to link a library in script editor is via `# r: `keyword.
Conda environments will be supported "maybe" in the future.

ChiaChing-Yen, 2024-05-08:
@petrasvestartas I followed this [thread](https://discourse.mcneel.com/t/new-python-editor-venv/165673/21) and it suggests adding `# env: /path/to/site-package` to the script to enable using that environment in Rhino. After adding this line, there's no need to use `# r: module`.
I used `conda` to create an environment with the same Python version as Rhino 8, then installed `compas` and `compas_fab` packages. I tried importing the modules, and they worked successfully.
While I haven't investigated this further, it seems like a viable workaround to make `conda` work with Rhino 8.

jf---, 2024-08-06:
I've run into the same issue: `compas` explicitly installs in the `ironpython` folder (`/.rhinocode/py27-rh8/Lib/site-packages` vs `/.rhinocode/py39-rh8/Lib/site-packages`, over `rhino3`, which seems at odds with the herculean effort of the compas team working towards `python3` compatibility.

```python
    # In Rhino 8 there is no scripts folder
    if version == "8.0":
        installation_path = compas_rhino._get_default_rhino_ironpython_sitepackages_path(version)
    else:
        installation_path = compas_rhino._get_rhino_scripts_path(version)
```

So @obucklin suggestion makes perfect sense to install into the `scripts` directory, so just factoring out the if statement should resolve things:

```python
In [6]: compas_rhino._get_rhino_scripts_path("8.0")
Out[6]: '/Users/jelle/Library/Application Support/McNeel/Rhinoceros/8.0/scripts'
```

In my view, this is a pressing issue, given that `compas` is getting increasing less / incompatible with `ironpython` and the considerable hinderance associated to running iron python (no numpy etc). To speak for myself: `python3` compatibility was more then sufficient reason to move to rhino 8 ;)

chenkasirer, 2024-08-15:
Sorry, I'm late for the party.
I made https://github.com/compas-dev/compas/pull/1384 and only then saw this thread after..

@tomvanmele could you please elaborate on point 1.? I don't quite get it.

I find installing compas to 3 places is necessary in Rhino8 as there seems to be 3 different interpreters (or 2 in 3 environments):
- Legacy IronPython (`Old` label)
- IronPython (this is both new GH IronPython components and the EditPythonCode diaglog in Rhino)
- CPython

And the corresponding locations are:
- `McNeel/Rhinoceros/8.0/scripts`
- `py27-rh8/Lib/site-packages`
- `py39-rh8/lib/site-packages`

I do agree we should utilize the `pip` mechanism that's now made possible but I had some weird issue happening when I tried it out with `compas_rhino.install_with_pip` plus I think it should eventually be integrated into the `compas_rhino.install`.

Another related issue I'm having is I cannot for the life of me import `compas_ghpython` in a python3 components. It doesn't matter if I install via pip or with symlinks. Just getting this persistent `NotSupportedInGHException`. It seems to be for some call related to layers but even removing all those I could find didn't get me anywhere. 

Any of you stumbled upon that or had better luck?

tomvanmele, 2024-11-15:
sure, but should we then not move that to a dedicated issue? currently, people read this as a page about problems with installing compas in Rhino, although it is only related to GH...

tomvanmele, 2024-11-15:
in any case, just a few thoughts...

### Rhino CPython (py39)

Basic users should use the default installation mechanism provided by Rhino, using `# r: compas, ...`.

```python
#! python3
# r: compas, ...
```

Different combinations of requirements for different projects can be isolated from each other using `# venv: ...`.

```python
#! python3
# venv: projectA
# r: compas, ...
```

More advanced users may want to/need to use `pip` directly themselves instead of using `# r: ...`. There are two options: use Rhino's `pip` or use the `pip` of a external (`conda`) environment with the correct `--target`. the target directive can also be combined with the usage of Rhino's `pip` to install packages into specific `venv`. not however, that Rhino adds a random string to the names of the virtual environments in `site-envs` (don't know why) and that you need to know what this string is in order to use `--target` properly.

This will install `compas` into the default site-env using Rhino's `pip`, directly from PyPI.

```bash
~/.rhinocode/py39-rh8/python3.9 -m pip install compas
```

This will install `compas` into a specific site-env ("projectA") using Rhino's `pip`, directly from PyPI.
The random string i used ("projectA-htl4ZgZ0") will be different for everyone, and needs to be looked up first.

```bash
~/.rhinocode/py39-rh8/python3.9 -m pip install compas --target ~/.rhinocode/py39-rh8/site-envs/projectA-htl4ZgZ0
```

The same can be done with source installs into the default or a specific site-env.

```bash
cd path/to/compas
~/.rhinocode/py39-rh8/python3.9 -m pip install .
```

```bash
cd path/to/compas
~/.rhinocode/py39-rh8/python3.9 -m pip install . --target ~/.rhinocode/py39-rh8/site-envs/projectA-htl4ZgZ0
```

Instead of using Rhino's `pip`, you can also use the `pip` of an external environment you might be working in (ideally this environment is based on Python 3.9, but depending on the requirements of the packages you install this is not always absolutely necessary). however, in this case `--target` is not optional, even for the default site-env.

```bash
conda activate compas-dev
cd path/to/compas
python -m pip install . --target ~/.rhinocode/py39-rh8/site-envs/projectA-htl4ZgZ0
```

Using an external environment directly as suggested above (using `# env: path/to/site-packages`) is not something i have been able to get to work, but i have only tried it on Mac. In any case, i doubt it will work for packages that are built and have build requirements like `compas_cgal`.

### Rhino IronPython

the "py27-rh8" folder has a very similar structure as the one of "py39-rh8" so i assume that at least the installation procedures listed above that use local source will work as well, but i haven't tried this...

### GH

About Grasshopper i have no idea, but i can confirm that Python 3 components have access to the same "site-envs" as the Rhino ScriptEditor.

[image]

chenkasirer, 2024-11-15:
That is my understanding of the situation as well.

We're basically retiring the `compas_rhino.install` mechanism, where, starting from Rhino8, you're expected to install compas inside Rhino as you described above. 
I find the instructions in https://compas.dev/compas/latest/userguide/cad.rhino8.html pretty clear and complete.

Perhaps we should remove the Rhino8 switch from `compas_rhino.install` so that it's not used as well as provide some kind of an uninstall script to allow removing COMPAS symlinks from `AppData\Roaming\McNeel\Rhinoceros\8.0\scripts` in case anyone has them there. This as a replacement for this PR https://github.com/compas-dev/compas/pull/1383 if @tetov agrees.

Once we've dealt with that, I'm fine with closing this issue.

**Regarding Grasshopper**

I think the way forward is components with `#r: ..` in header and that are distributed as YAK packages over Food4Rhino.
We'd probably need to separate the IPY and CPython component definitions and make some changes to `compas_invocation2` as well as integrate that to the componentizer action.

We're currently trying this out with COMPAS Timber. still some things that used to work differently in Rhino7 we're struggling with. will update.

For completeness sake I'll just mention that this `#r: ...` syntax doesn't quite work for me in the IronPython components in Rhino8. maybe I'll post in the mcneel forum.

Finally there's the legacy ironpython components which perhaps we shouldn't support.

Maybe we can post updates about the Grasshopper situation somewhere.

chenkasirer, 2024-11-15:
One open question for me is how to continue supporting Rhino7 Grasshopper component installation. If we continue installing the user objects to the same place we do now (`AppData\Roaming\Grasshopper\UserObjects`) the components appear in both Rhino7 and Rhino8..

Evidently, you can get Grasshopper plugins for Rhino7 over Food4Rhino, but how they manage python library dependencies I have no idea.

tomvanmele, 2024-11-16:
it sounds like `AppData\Roaming\Grasshopper\UserObjects` is more or less the equivalent of the scripts folder. can we not install directly into `AppData\Roaming\McNeel\Rhinoceros\packages\7.0`? that way 7.0 and 8.0 are cleanly separated, which allows us to use different installation and dependency management approaches in both cases...

chenkasirer, 2024-11-19:
Agree.
to sum up and make actionable I propose the following (I can make the changes)
* `compas_invocations2` add ability to distinguish between component source for ironpython and cpython (as these will most likely need to co-exist)
* `compas-actions.ghpython_components` add the `yakerize.py` script (thanks @9and3!) to the componentizer and expose it via the github action
* `compas_invocations2` add `yakerize` invoke command to allow local packaging
* `compas` - remove the `8.0` flag from `compas_rhino.install`
* `compas` - add cpython compatible copy of the components to `compas_ghpython\components_cpython`
* `compas-actions.build/publish` - extend to build yak packages for rhino7 and rhino8

when using YAK components are installed to `AppData\Roaming\McNeel\Rhinoceros\packages\7.0` and `...\8.0` 
the Rhino8 components need to have `#r: compas` at the top which will take care of the python dependencies.

Rhino7 components would need the good old `compas_rhino.install -v7.0` to have their dependencies available.

tomvanmele, 2024-11-20:
@tetov ping

tomvanmele, 2024-11-20:
@chenkasirer just fyi... 

when releasing the new version of RhinoVault through Yak, we noticed that, at least on the Rhino side, you have to be careful with the requirements inside the command scripts. for some strange reason, even if there are no conflicts between versions, having `compas` listed alongside other requirements which in turn listed `compas` as one of their requirements in their `requirements.txt` made the commands reinstall all requirements every time they were run. once we removed `compas` from the list, the problem went away...

the problem didn't occur as long as we were using locally build packages.

for example, a script with `# r: compas, compas_tna, compas_session` would trigger this behaviour.

however, the same script with `# r: compas_tna, compas_session` would not...

ricardoavelino2, 2025-02-04:
> not sure if they are helpful, but here are a few thoughts...
> 
> ### Rhino CPython (py39)
> Basic users should use the default installation mechanism provided by Rhino, using `# r: compas, ...`.
> 
> #! python3
> # r: compas, ...
> Different combinations of requirements for different projects can be isolated from each other using `# venv: ...`.
> 
> #! python3
> # venv: projectA
> # r: compas, ...
> More advanced users may want to/need to use `pip` directly themselves instead of using `# r: ...`. There are two options:
> 
> * use Rhino's `pip`
> * use the `pip` of a external (`conda`) environment with the correct `--target`.
> 
> the target directive can also be combined with the usage of Rhino's `pip` to install packages into specific `venv`. not however, that Rhino adds a random string to the names of the virtual environments in `site-envs` (don't know why) and that you need to know what this string is in order to use `--target` properly.
> 
> **Rhino pip**
> 
> Install `compas` into the default site-env using Rhino's `pip`, directly from PyPI:
> 
> ~/.rhinocode/py39-rh8/python3.9 -m pip install compas
> Install `compas` into a specific site-env ("projectA") using Rhino's `pip`, directly from PyPI: (The random string i used ("projectA-htl4ZgZ0") will be different for everyone, and needs to be looked up first.)
> 
> ~/.rhinocode/py39-rh8/python3.9 -m pip install compas --target ~/.rhinocode/py39-rh8/site-envs/projectA-htl4ZgZ0
> The same can be done with source installs into the default or a specific site-env:
> 
> cd path/to/compas
> ~/.rhinocode/py39-rh8/python3.9 -m pip install .
> cd path/to/compas
> ~/.rhinocode/py39-rh8/python3.9 -m pip install . --target ~/.rhinocode/py39-rh8/site-envs/projectA-htl4ZgZ0
> **NOT Rhino pip**
> 
> Instead of using Rhino's `pip`, you can also use the `pip` of an external environment you might be working in (ideally this environment is based on Python 3.9, but depending on the requirements of the packages you install this is not always absolutely necessary). however, in this case `--target` is not optional, even for the default site-env.
> 
> conda activate compas-dev
> cd path/to/compas
> python -m pip install . --target ~/.rhinocode/py39-rh8/site-envs/projectA-htl4ZgZ0
> Using an external environment directly as suggested above (using `# env: path/to/site-packages`) is not something i have been able to get to work, but i have only tried it on Mac. In any case, i doubt it will work for packages that are built and have build requirements like `compas_cgal`.
> 
> ### Rhino IronPython
> the "py27-rh8" folder has a very similar structure as the one of "py39-rh8" so i assume that at least the installation procedures listed above that use local source will work as well, but i haven't tried this...
> 
> ### GH
> About Grasshopper i have no idea, but i can confirm that Python 3 components have access to the same "site-envs" as the Rhino ScriptEditor.
> 
> <img alt="Screenshot 2024-11-15 at 14 24 55" width="945" src="https://private-user-images.githubusercontent.com/29653489/386617532-4054cce2-e3e1-4f86-bda1-5ec7e70156e3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzg2MzI4OTUsIm5iZiI6MTczODYzMjU5NSwicGF0aCI6Ii8yOTY1MzQ4OS8zODY2MTc1MzItNDA1NGNjZTItZTNlMS00Zjg2LWJkYTEtNWVjN2U3MDE1NmUzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjA0VDAxMjk1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2NDhjN2JjN2JiNmQ0NjIwNDQ0ZGQ3YTYwNzBhOWJhMmVhOGNmZDk5YzdkOGRlODA5MTAwNzlhMjVjM2JkMjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.8SMEvjT0lWnUc1T1sx8aDaM--Zxe4J9_Hddnx5AkNDs">

This should go into the Rhino 8 install manual

tetov, 2025-02-04:
@ricardoavelino2:

Updated now I think: https://compas.dev/compas/latest/userguide/cad.rhino8.html