compas-dev/compas #1354: `Runtime error (NotSupportedException)` in Grasshopper when using `draw_labels`

chenkasirer, 2024-05-13:
Getting a runtime error when trying to draw a text label in Rhino with Grasshopper.

Can be reproduced by running the following code in a GHPython component:
```python
from compas_rhino.utilities import draw_labels

draw_labels([{"pos": [0, 0, 0], "text": "this is a text"}])
```
The full error:
```
Runtime error (NotSupportedException): This type of object is not supported in Grasshopper, so this Python script cannot create it. You might want to use 'scriptcontext.doc = Rhino.RhinoDoc.ActiveDoc' to use the Rhino doc, instead? If you do, remember to restore it: 'scriptcontext.doc = ghdoc'.

Traceback:
  line 110, in draw_labels, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas_rhino\utilities\drawing.py"
  line 57, in wrapper, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas_rhino\utilities\drawing.py"
  line 3, in script
```
**Desktop (please complete the following information):**
 - OS: Window 7
 - Rhino 7 SR36

**Additional context**
The suggested solution in the error (setting `scriptcontext.doc` to `RhinoDoc.ActiveDoc`) doesn't produce a different result.
However, this code works:
```python
import Rhino

text = Rhino.Geometry.TextDot("this is a text", Rhino.Geometry.Point3d(0, 0, 0))
Rhino.RhinoDoc.ActiveDoc.Objects.AddTextDot(text)
```

[image]

tomvanmele, 2024-05-13:
i guess the question is if GH components should create baked text dots...

chenkasirer, 2024-05-14:
Ahh that's a good point. Was so annoyed at this not working (was the support removed? I vaguely recall this used to work..) that I didn't even notice it's baked..