compas-dev/compas #1355: RobotModel SceneObject with Python3 in Rhino8

gidodielemans, 2024-05-14:
**Describe the bug**
Robot visualization not possible with Python3 in Rhino8

**To Reproduce**
Steps to reproduce the behavior:
`from compas_robots.ghpython.scene import RobotModelObject

# Load URDF from ROS
robot = ros_client.load_robot(
            load_geometry=True, 
            urdf_param_name= '{}/robot_description'.format(prefix),
            srdf_param_name= '{}/robot_description_semantic'.format(prefix))
robot.scene_object = RobotModelObject(robot.model)`

**Expected behavior**
A RobotModelObject for visualization is attached to the Robot object

**Error messages**
No scene object is registered for this data type: <type 'compas_robots.robot.model.RobotModel'>
No module named System.Array

**Desktop (please complete the following information):**
 - OS: Windows
 - Python 3.9.10
 - Python package manager Rhino (.rhinocode/py3.9-rh8)

gidodielemans, 2024-05-14:
#1356 Here an attempt on a fix for the System module incompatibility

tomvanmele, 2024-05-14:
hi, i think this is meant for `compas_robots`. there is no `RobotModelObject` in `compas` itself...

tomvanmele, 2024-05-14:
yes, there are quite a few of these `System` problem in `compas_ghpython` in Rhino 8. thanks for the fix. will have a look...

gidodielemans, 2024-05-15:
Related pull request in compas_robots https://github.com/compas-dev/compas_robots/pull/16

closed 2024-09-27 by gidodielemans