compas-dev/compas #1359: Transformation of geometry

Licini, 2024-05-24:
```python
from compas.geometry import Box
from compas.geometry import Scale

box = Box.from_width_height_depth(1, 1, 1)
print(box)

box.transform(Scale.from_factors([2, 2, 2]))
print(box)

box.scale(2)
print(box)

from compas.geometry import Sphere

sphere = Sphere(1)
print(sphere)

sphere.transform(Scale.from_factors([2, 2, 2]))
print(sphere)

sphere.scale(2)
print(sphere)
```

```
Box(xsize=1.0, ysize=1.0, zsize=1.0, frame=Frame(point=Point(x=0.0, y=0.0, z=0.0), xaxis=Vector(x=1.0, y=0.0, z=0.0), yaxis=Vector(x=0.0, y=1.0, z=0.0)))
Box(xsize=1.0, ysize=1.0, zsize=1.0, frame=Frame(point=Point(x=0.0, y=0.0, z=0.0), xaxis=Vector(x=1.0, y=0.0, z=0.0), yaxis=Vector(x=0.0, y=1.0, z=0.0)))
Box(xsize=2.0, ysize=2.0, zsize=2.0, frame=Frame(point=Point(x=0.0, y=0.0, z=0.0), xaxis=Vector(x=1.0, y=0.0, z=0.0), yaxis=Vector(x=0.0, y=1.0, z=0.0)))
Sphere(radius=1.0, frame=Frame(point=Point(x=0.0, y=0.0, z=0.0), xaxis=Vector(x=1.0, y=0.0, z=0.0), yaxis=Vector(x=0.0, y=1.0, z=0.0)))
Sphere(radius=1.0, frame=Frame(point=Point(x=0.0, y=0.0, z=0.0), xaxis=Vector(x=1.0, y=0.0, z=0.0), yaxis=Vector(x=0.0, y=1.0, z=0.0)))
Sphere(radius=2.0, frame=Frame(point=Point(x=0.0, y=0.0, z=0.0), xaxis=Vector(x=1.0, y=0.0, z=0.0), yaxis=Vector(x=0.0, y=1.0, z=0.0)))
```

We need to be careful about this... I understand this is tricky because of non-uniform scale, maybe there should a some kind of warning saying the Scale transform might not work as expected

gonzalocasas, 2024-05-24:
It's kind of broken now, right? `scale` vs `scaled` do different things also

tomvanmele, 2024-05-24:
yes, `scaled`, `translated`, `rotated` don't use the corresponding in-place version but rather a modified version of `transformed`.

i can fix this...

tomvanmele, 2024-05-26:
#1360