compas-dev/compas #1389: Problem with Mesh.face_circle()

robin-oval, 2024-09-10:
The `Frame` in `Circle` in `Mesh.face_circle()` is wrongly initiated as `Circle((point, normal), radius)`, instead of `Circle(radius, (point_0, point_1, point_2))`.

robin-oval, 2024-09-10:
> @robin-oval <https://github.com/robin-oval>
>
> Did you check the constructor overloads too?
>
> from_plane_and_radiusfrom_point_and_radiusfrom_three_pointsfrom_points

I saw them, but it seems that `Mesh.face_circle()` uses the base
constructor, not `from_plane_and_radius`.

petrasvestartas, 2024-09-10:
I also get the same error, the circle constructor needs to be changed to this:

```python
    def face_circle(self, face):
        """The circle of a face.

        Parameters
        ----------
        face : int
            The face identifier.

        Returns
        -------
        :class:`compas.geometry.Circle`
            The circle of the face.

        """
        from compas.geometry import bestfit_circle_numpy
        point, normal, radius = bestfit_circle_numpy(self.face_coordinates(face))
        return Circle.from_plane_and_radius(Plane(point, normal), radius)
```

instead of
`
Circle((point, normal), radius)`

[image]

@robin-oval  can you make a pull-request?

robin-oval, 2024-09-11:
Solved with PR

closed 2024-09-11 by robin-oval