compas-dev/compas #1411: unify_cycles / face_adjaccency

romanarust, 2024-11-14:
**Describe the bug**

I have a set of [vertices and faces](https://gist.githubusercontent.com/romanarust/58e5dfea928ecd2207dd21d5b512ba4e/raw/7c0cad520963c2d2fa013e0b9d48285934de23d7/vertices_faces.json) from a correct, closed mesh, but the function `unify_cycles` fails. The reason for this is that when there are more than [100 faces](https://github.com/compas-dev/compas/blob/46b286311ccf0212f65fa202684c978a1d6bf560/src/compas/topology/orientation.py#L122), a different function (`_face_vertices`) is used for face adjacency. However, in my case, this function produces a different, and wrong result than `face_adjacency`, because the parameters of this function aren’t dependent on the input.

How would it be best to proceed: I could propose a fix, but should the parameters be determined based on the input (i.e determine the radius on the longest edge?), or should the user be allowed to select which function to use (`unify_cycles(fast=True)`)? My understanding is that the `_face_adjacency` function was intended to increase performance. Wouldn’t it make sense for this to be user-controlled?

tomvanmele, 2024-11-14:
hi romana,

is `radius=10` the source of the problem?

tomvanmele, 2024-11-14:
perhaps both?

* user is allowed to choose, with the default switch for over 100 faces
* radius is None by default and in that case is computed based on some kind of heuristic
* radius is available as a top-level parameter so it can be set by the user in case of awkward mesh geometies

romanarust, 2024-11-14:
ok, I'll propose something