compas-dev/compas #1419: Error: compas.set_precision

nizartaha, 2025-01-10:
compas.set_precision(0.001)
AttributeError: module 'compas' has no attribute 'set_precision'

https://compas.dev/compas/1.5.0/api/generated/compas.set_precision.html

I wanted to round the Frame values to 6 digits  , how is it possible when using `compas.json_dump` ?

```
 "data": {
    "point": [
        665.6414239488341,
        -218.07475329238576,
        652.9999999999999
    ],
    "xaxis": [
        0.25779917216805537,
        0.9661985235082206,
        -6.938893903907227e-18
    ],
    "yaxis": [
        0.9661985235082206,
        -0.25779917216805537,
        -1.224646799147353e-16
    ]
},
"dtype": "compas.geometry/Frame",
"guid": "77685a15-8823-4110-8eaf-1aa445e233b8"
```

gonzalocasas, 2025-01-10:
@nizartaha Hi Nizar, are you using compas 1.5? On newer COMPAS releases (2.x+), this was moved to the [tolerance module](https://compas.dev/compas/latest/api/compas.tolerance.html). However, I think the storage to json will not round things, it will preserve original precision, but the precision will be used for comparisons and other geometric operations. Changing it in the json would be a destructive operation, if you really want to round the data, you will need to do something more explicitly.  (@tomvanmele correct me if I'm wrong here, please)

nizartaha, 2025-01-10:
Hi @gonzalocasas good point. i am using `2.4.2` (my fault, the docs was open for `1.5` :/)

I see , if it has to be done explicitly then would it be after the data is exported? what do you suggest?

gonzalocasas, 2025-01-29:
> I see , if it has to be done explicitly then would it be after the data is exported? what do you suggest?

To do it explicitly, you need to actively round up before you export. I personally think that's a bad idea. Data should be preserved untouched on disk. Rounding can always happen at time of usage/display. Otherwise, loading and saving multiple times the same data could potentially end up creating larger differences due to the fact that data needs to go through multiple different representations that don't preserve exact information.

gonzalocasas, 2025-01-29:
@nizartaha maybe we can move this to the forum, or continue the conversation here, but close the issue, because I don't think this is actionable on our side. What do you think?

nizartaha, 2025-01-29:
@gonzalocasas Sure, Thanks

closed 2025-01-29 by nizartaha