compas-dev/compas #1447: Brep plugin alignment with interface

chenkasirer, 2025-03-20:
Made a [script](https://gist.github.com/chenkasirer/2e9e396286109753f63446edeee0810f) comparing the level of interface implementation of out brep plugins.

@tomvanmele perhaps we could use this as reference in effort to better align the plugins. Mostly, it seems, RhinoBrep is missing a bunch of stuff, so I'll get started on those.

```bash
####################################################################################################
#### OCCBrep Analysis ####
####################################################################################################

    ################################################################################
    ## Constructors missing in OCC ##
    ################################################################################

    - from_loft
    - from_pipe

    ################################################################################

    ################################################################################
    ## Properties missing in OCC ##
    ################################################################################

    - curves
    - is_valid
    - surfaces
    - trims

    ################################################################################

    ################################################################################
    ## Methods missing in OCC ##
    ################################################################################

    - contains
    - edge_loop
    - flip

    ################################################################################

    ################################################################################
    ## Methods not in Interface ##
    ################################################################################

    - check
    - copy
    - edge_loops
    - simplify
    - transform
    - transformed

    ################################################################################

    ################################################################################
    ## Properties not in Interface ##
    ################################################################################

    - occ_shape

    ################################################################################

####################################################################################################
#### RhinoBrep Analysis ####
####################################################################################################

    ################################################################################
    ## Constructors missing in Rhino ##
    ################################################################################

    - from_brepfaces
    - from_breps
    - from_cone
    - from_curves
    - from_iges
    - from_pipe
    - from_plane
    - from_planes
    - from_polygons
    - from_sweep
    - from_torus

    ################################################################################

    ################################################################################
    ## Properties missing in Rhino ##
    ################################################################################

    - centroid
    - curves
    - is_closed
    - is_compound
    - is_compoundsolid
    - is_convex
    - is_infinite
    - is_orientable
    - is_shell
    - is_surface
    - is_valid
    - orientation
    - shells
    - solids
    - surfaces
    - type

    ################################################################################

    ################################################################################
    ## Methods missing in Rhino ##
    ################################################################################

    - contours
    - cull_unused_edges
    - cull_unused_faces
    - cull_unused_loops
    - cull_unused_vertices
    - edge_faces
    - edge_loop
    - fix
    - heal
    - make_solid
    - overlap
    - sew
    - to_iges
    - to_polygons
    - to_step
    - to_stl
    - to_tesselation
    - to_viewmesh
    - vertex_edges
    - vertex_faces
    - vertex_neighbors

    ################################################################################

    ################################################################################
    ## Methods not in Interface ##
    ################################################################################

    - copy
    - transform

    ################################################################################

    ################################################################################
    ## Properties not in Interface ##
    ################################################################################

    ✔ No missing elements.

    ################################################################################
```

jf---, 2025-03-20:
> comparing the level of interface implementation of our brep plugins

terrific idea!

obucklin, 2025-03-24:
I think @papachap and I will help with this when we can. How do we "claim" functions so we don't do things twice?

tomvanmele, 2025-03-24:
create an issue that has the function in the title?

chenkasirer, 2025-04-16:
Adding missing Rhino properties and methods:

**Alternative constructors**
- [ ] from_brepfaces
- [ ] from_breps
- [ ] from_cone
- [ ] from_curves  - https://github.com/compas-dev/compas/issues/1451
- [ ] from_iges
- [ ] from_pipe
- [ ] from_plane
- [ ] from_planes
- [ ] from_polygons
- [ ] from_sweep
- [ ] from_torus

**Properties**
- [ ] centroid
- [ ] curves
- [ ] is_closed
- [ ] is_compound
- [ ] is_compoundsolid
- [ ] is_convex
- [ ] is_infinite
- [ ] is_orientable
- [ ] is_shell
- [ ] is_surface
- [ ] is_valid
- [ ] orientation
- [ ] shells
- [ ] solids
- [ ] surfaces
- [ ] type

**Methods**
- [ ] contours
- [ ] cull_unused_edges
- [ ] cull_unused_faces
- [ ] cull_unused_loops
- [ ] cull_unused_vertices
- [ ] edge_faces
- [ ] edge_loop
- [ ] fix
- [ ] heal
- [ ] make_solid
- [ ] overlap
- [ ] sew
- [ ] to_iges
- [ ] to_polygons
- [ ] to_step
- [ ] to_stl
- [ ] to_tesselation
- [ ] to_viewmesh
- [ ] vertex_edges
- [ ] vertex_faces
- [ ] vertex_neighbors