compas-dev/compas #1468: Proposal: Scene objects should not be replaced during the plugin phase, only the functionality of methods

tomvanmele, 2025-05-29:
Currently, core scene objects are replaced during the plugin phase by their implementation context counterparts. This leads to all sorts of problems.

* the API of the context object implementation may or may not match the expected API of the documented default implementation
* switching context of the scene is very difficult
* objects behave differently in different contexts (because of lack of control over the provided API)

By making only the functionality of scene object methods pluggable, instead of replacing the entire object, at least some of these problems go away.

gonzalocasas, 2025-05-30:
Isn't it going to mean that there's a huge set of functions to implement as plugins?

> the API of the context object implementation may or may not match the expected API of the documented default implementation

I doubt this will improve if the implementations are function plugins instead, it's still the same amount of features that need to be implemented, and we don't have a clean overview of what the must-have API surface needs to be.

I'm not exactly against this change, but I think this can be solved with better documentation (maybe automated parsing, like what @chenkasirer did a while ago), rather than technical changes

jf---, 2025-06-26:
@tomvanmele can you give an example of some of the aspects you've raised? this seems a pressing issue.